LilyGO/LILYGO-T-Energy #1: T18_3.0 Schematic

lacrimal, 2019-12-12:
Hi,
Can you publish schematic for T10_3.0 20190410 ?
Thanks

immertroll, 2020-01-18:
Yeah that would be really nice. It would be also good if you could add some description.
I like to know if one pin is connected internally to measure the battery voltage a.s.o.
Thx

vladsol, 2020-02-15:
I'm also interested :)

kapitan-u, 2020-02-27:
Please publish schematics for T18_3.0 20190410 !
I'm wondering if there is any connections for measuring battery voltage.

meditant, 2020-04-02:
Me too, if you can publish schematics and if it's have the same measuring battery voltage like the T8 2.2 Lion Oled

Best,

GearWalker, 2020-05-10:
Please, release it!

gexpo, 2020-11-18:
https://github.com/LilyGO/LILYGO-T-Energy/blob/master/WiFiScan/WiFiScan.ino

xeijin, 2020-11-28:
I take it this isn't the schematic? (Genuine question, I am just starting out and also looking for it)

https://github.com/LilyGO/LILYGO-T-Energy/blob/master/t18_v3.0.pdf

gexpo, 2020-11-30:
thank you.
regards.
gennaro

??

https://github.com/LilyGO/LILYGO-T-Energy/blob/master/t18_v3.0.pdf

tedder, 2021-07-06:
[from the schematic](https://github.com/LilyGO/LILYGO-T-Energy/blob/master/t18_v3.0.pdf), battery voltage is on a voltage divider (100k/100k) on pin 35.

LED is at pin 5.

The BME280 isn't populated, but it would be at pins 20/21 for i2c.

tedder, 2021-07-27:
and here's a pinout from lilygo.cn:
https://ae01.alicdn.com/kf/HTB1PTD2TrrpK1RjSZTEq6AWAVXay.jpg

[image]