LilyGO/LILYGO-T-Energy #7: Built in battery voltage sensor.

SyvelDesign, 2021-05-31:
Hello,

I've been told that there is a built in battery voltage sensor on pin36. Is that correct? No matter what I try I can't read it.

Thanks!

AlekseyFedorovich, 2021-06-15:
I've seen many interested people in the issues here but no definitive answer: is there any built-in mechanism for measuring own battery level? I could figure it out from the schematics

SyvelDesign, 2021-06-15:
I figured it out from reading the schematics. It reports on pin 35, not pin 36 like I was told. Then if you apply some simple math to it like ... (measurement / 4095.0) * 7.26 ... you get a (probably not super accurate) battery voltage reading.

AlekseyFedorovich, 2021-06-15:
what is the 7.26 factor for?

SyvelDesign, 2021-06-15:
From another post about reading voltage on a ttgo t-display.

"float battery_voltage = ((float)v / 4095.0) * 2.0 * 3.3 * (1100 / 1000.0);

So 1100 is actually 1.1V in mV. 1.1V is the internal reference voltage that readings get compared to.

Then 2.0 * 3.0 are voltage divider parameters. You can't measure 5V on an ADC with a 1.1V reference voltage you can only measure from 0V to 1.1V. That's why they used voltage dividers to extend the range to 2.0 * 3.3 * 1.1 = 7.26V.
You actually have to set the reference voltage by setting the right attenuation adc1_config_channel_atten(ADCx_CHANNEL_x, ADC_ATTEN_0db); Something like this.

Then v / 4095.0 is there because ESP32 has a 12bit ADC and 12bits is 4096 steps, so 0V is 0 and 1.1V is 4095. When you multiply that with 7.26, you get a range where 0 on ADC is 0.0V and 4095 is 7.26V."

Hopefully that helps explain a little.

MrMcChip, 2021-11-13:
And it's simplest form a potentiometer wired from 0 to 3.3 volts with its wiper connected on a analog input will generate the full range of binary count for 12 bits that being 0 volts is 0 binary and 3.3 volts being 4095 binary

Zixim, 2021-12-28:
Could someone please post the code for reading VBAT ?
Also, is anyone using ESPHome on this board ?

theloukou, 2022-03-21:
To get your voltage just analogRead pin 35. Simple math and you get the read voltage.
The included divider is set to give you half battery voltage. Whatever the ADC puts out, multiply by 2. it's dead simple really...

@Zixim I am actively using this board with ESPHome, it's just another ESP32.

kmilokeepex, 2022-11-17:
here is the code that work for me to reed the batery voltage.

// Bateria is connected to GPIO 35 (Analog ADC1_CH6) 
//const int batPin = 35;
#define PIN_ADC_BAT 35
#define ADC_BATTERY_LEVEL_SAMPLES 50
void read_adc_bat(uint16_t *voltage);
uint16_t v_bat = 0;

void read_adc_bat(uint16_t *voltage)
{
  uint32_t in = 0;
  for (int i = 0; i < ADC_BATTERY_LEVEL_SAMPLES; i++)
  {
    in += (uint32_t)analogRead(PIN_ADC_BAT);
  }
  in = (int)in / ADC_BATTERY_LEVEL_SAMPLES;
  uint16_t bat_mv = ((float)in / 4096) * 3600 * 2;
  *voltage = bat_mv;
}

void setup() {
  pinMode(PIN_ADC_BAT, INPUT); 
  read_adc_bat(&v_bat);
  Serial.print("BAT: ");
  Serial.println(v_bat);
}

BojanJurca, 2024-07-13:
Hi, according to https://www.tinytronics.nl/en/development-boards/microcontroller-boards/with-wi-fi/lilygo-ttgo-t-energy-esp32-wrover-with-18650-battery-holder TTGO T-Energy ESP32-WROVER measures battery voltage on pin 35, version 3.0 already has a built in voltage divider made of two 100K resistors.

the following code works for me:

```
uint16_t raw = (int16_t) (~adc1_get_raw (ADC1_CHANNEL_7) & 0xFFF); // read GPIO 35, the bits are inverted
float v = raw * 3.600 / 4095.0 * 2;
Serial.printf ("raw: %i, Vbat: %f\n", raw, v);
```