LilyGO/LILYGO-T-Energy #12: T18_V3.0 Board doesn't wake up from timer deep sleep

sle118, 2023-07-13:
The schematics for this board show what seems to be a FET which can be used to shut down the 3.3V power, hooked to GPIO26. Can anyone confirm? The lack of component identification on this board is just aggravating the issue as it makes troubleshooting with a scope very difficult.

I'm scratching my head over deep sleep, since the board never actually comes back up as it should.

sle118, 2023-07-13:
well... nevermind this, there was possibly a software issue with my code. It works now so I'll just go ahead and close this issue

sle118, 2023-07-17:
I just wanted to comment on this as, for some obscure reason that I haven't figured out yet, the board will wake up several times and all of a sudden just decide to stop waking up. It doesn't look like the issue comes from the battery, as hard resetting the device with the power switch revives it for more sleep cycles with the battery voltage continuing its drop to 3.7V (last time I revived the board).

I'm puzzled with this one, in case anyone has some advice.

sle118, 2023-07-21:
ok. for anyone interested, here's the catch for this. I had a variable sleep duration that came from the homeassistant platform (through the API) and the value wasn't always updated after a reboot from deep sleep. So I turned to storing the value to flash when it changes (global variable in ESPHOME with persistance) and the wakeup issue is vanished.

now this freaking board is just killing batteries and a 18650 cell won't make much more than 30 hours at best. The starting idea was good, but the design is poor and disapointing. 

Why is no one from @LilyGO even responding to any of the issues in this repository?

Verdagon, 2025-03-16:
I'm also seeing some issues with sleep modes... calling `esp_light_sleep_start` or `esp_deep_sleep_start` both crash immediately, no matter what I do to set it up (calling `esp_sleep_enable_timer_wakeup`, etc.)

Even their sample at https://github.com/Xinyuan-LilyGO/T-Energy-S3/blob/main/examples/Deep_Sleep_Wake_Up/Deep_Sleep_Wake_Up.ino crashes on the `esp_deep_sleep_start` call.

@sle118 I assume you never found a way to make deep sleep actually work?

Keivan45, 2025-04-25:
We conclude that don't buy Chinese goods because they have no use except throwing them in the trash.